UVicFH/Home #3: On-boarding Discussion

BrendonEarl, 2017-05-19:
Discuss on-boarding procedure with the team.
If you're coming think about 'your version' of what on-boarding should look like and be ready to discuss.

benjamin-elk, 2017-05-25:
Some ideas:
-get all new recruits to do some sort of online survey that will give us an understanding of their areas of interest, experience, projects they would like to work on, etc..

-send new recruits an email that formally welcomes them to the team, gives them some key resources, like how to join our slack channel, where to find the FH rules, any other info that seems pertinent

-start by giving new recruits a small projec to gauge their interest (how quickly they get the project done, the quality of their work). Important: spell out what is expected of them: timeline, results, written report

RAKuhr, 2017-05-25:
We will be having Amal (she was the previous on-boarder for the team) and McKenzie Nicks (who I think would be well suited to the role of on-boarder this season) join us remotely.

Ideally we should focus on goals/objectives of the process and discuss the scope of the role so that everyone is clear where responsibilities begin and end.

RAKuhr, 2017-05-25:
Agenda:

1 Introduce McKenzie and Amal 
1. We will open up with a review of the previous process.
2. Amal will discuss her experiences and make any recommendations (she is 3h ahead so we want to make sure she can bust out when she needs to go to bed)
3. Goal setting as a group.
4. Open the floor for individual input (limit of 5 min per individual)
5. Snack break (15 min)
6. Open discussion (30 min)
7. Determine responsible parties for follow-up.

Let me know if anyone wants to add or make modifications

smithbeuvic, 2017-05-26:
minutes from may 25 onboarding meeting 

in the past:
amal would keep 6 weeks steady contact
main focus was on the students who were not super driven/ " on in betweeners "-Amal
main factor causing loss of potentially good team members was lack of project availability combined with schoool comittments 

what worked:
tutorials, ie. Solidworks and Altium

Recommended by Amal:
1.) identify types of candidates...vulnerability of lost potential
2.) allocate resources towards the new members that have potential and are at risk of being lost
3.) people need to know what they can do/ how to be helpefull

SOMETHING PRESENTED BY BEN DURING MEETING
https://docs.google.com/document/d/15oTptUYfOBULG_EEXDvkjskwfhoy-LwZ4qK8pzGpJq4/edit?usp=sharing

amal is sending information to mkcenzie