UVicFH/Home #5: Preliminary Chassis Design Research & Objectives

MrMikeGrant, 2017-05-27:
### Context

For the 2018 racing season, the UVic Formula Hybrid team will be designing and fabricating a new chassis. As this is a major project that has a significant influence on the cars performance it must be researched thoroughly to ensure the final design is optimized for the team's objectives.

### Outcomes

- [ ] Read chassis design texts specifically geared towards racing suspensions (2-3 minimum)

- [ ] Gain a solid understanding of the chassis design process and industry best practices

- [ ] Develop objectives and goals for the new chassis design 

### Constraints

N/A

### Related Issues
N/A

### Steps to reproduce behavior

N/A

MrMikeGrant, 2017-05-27:
@smithbeuvic Please add a list of the chassis design texts you are reading and where team members can located them on (i.e. Google Drive). It would be great if you could order them from most critical to least critical!

smithbeuvic, 2017-05-27:
ok ;), not a professional yet so it will be my opinion

smithbeuvic, 2017-06-02:
we still need to get all the components measured and modelled

closed 2017-06-27 by smithbeuvic