UVicFH/Home #6: GitHub Workflow

BrendonEarl, 2017-05-27:
### Context

Public place to talk about how we work with GitHub

### Outcomes

A document outlining how to work with our GitHub issue handling

### Current (Basic) way of handling

* Create Issue
* Assign time label
* Assign severity label
* Assign status label
* Assign type label
* Assign person to issue if necessary

BrendonEarl, 2017-05-27:
In particular I'd like to discuss how we can use milestones and projects to group and organize issues without having duplication or too many places to look

chadjmccolm, 2017-06-06:
Created the template for the main page in the Wiki and an example project page. The example project page was **not** completed because I couldn't recall the exact format and the SMUS kids are still in the shop so I can't go reference the whiteboard. Hopefully somebody else can chime in and remember what we decided on or go take a picture of the whiteboard.

BrendonEarl, 2017-06-06:
On the main page: this years projects at the top with [in progress] tags as necessary, and sections for each year (each linking to changed projects)
Then in each project top is summary, for each year (with the newest at the top): justification, process, a few other headers, and future changes.

BrendonEarl, 2017-12-05:
@chadjmccolm @smithbeuvic Hey guys - just noticing the naming of some of the repo's on here are in the form of <year>-<project-name> or <year-range>-<projectName>. What do you think of keeping them as just <projectName> (in CamelCase) since GitHub takes care of versioning & time stamping for us?