UVicFH/Home #10: Powertrain Meeting

chadjmccolm, 2017-06-25:
### Background
A meeting was held June 24, 2017 with those interested in working on powertrain projects and development. The goal of this meeting was to identify interested members and work out what our individual goals were in order to develop the goals of the team.

### Attendance
Chad McColm (discussion leader)
Benjamin Kosten
Kurt Elliott
Ciara Duesterwald

### Outcomes
We decided that our primary short-term goal was developing knowledge about the current systems through inspection and experimentation. Our other goals were as follow: 
- Catch up on all overdue or unknown maintenance
    - Clean Air Filter
    - Clean Fuel Filter
    - Clean Injectors
    - Change Spark Plug
    - Check Oil Pressure Reg. Valve
    - Change Clutch Hydraulic Fluid
- Remove the engine two weeks from now if the car isn't running by then
- Develop a voltage divider circuit that will be safe to use as the ETB signal
    - Between 0-5V 
    - Releases if it is dropped or let go
- Plan a thermostat project

### Further Actions
Further actions include: 
- Designating who will design the thermostat
- Designating who will design the voltage divider
- Contacting previous powertrain group to see if any maintenance items above have been done but haven't been recorded

isaaccormack, 2017-06-25:
A thermostat was designed last year, but not fabricated nor implemented. The CAD can be seen on GrabCAD - "0002-R1-Thermostat". Obviously if you want to take the design another way go ahead, just letting you know.

isaaccormack, 2017-06-25:
What is the estimated time to complete for the voltage divider project? That sounds like fun.

chadjmccolm, 2017-06-26:
@isaaccormack I would like to have it done in three weeks (by the 15th). Estimated time would be < 10 hours. We'd also need to double check that this is the only externally signalling wire (I'm like 80% sure it is).

At minimum it can be a 3 hour project with a small breadboard, a battery, a resistor, a slide potentiometer, and some sort of kill switch (depending on what we have in stock - emergency stop buttons can be surprisingly expensive).

chadjmccolm, 2017-07-02:
We are taking the thermostat as a opportunity to redesign our cooling system to accommodate the new radiator. This is issue #11. Our goal time to complete the cooling system design work is Tuesday afternoon.

chadjmccolm, 2017-09-14:
Powertrain has become a diversified issue that we're breaking into chunks following the learning in the [engine review doc](http://goo.gl/YWCBNp). This issue will be closed and more developed issues will be created.

closed 2017-09-14 by chadjmccolm