UVicFH/Home #11: Cooling System Redesign

chadjmccolm, 2017-07-02:
### Context

We need a thermostat and housing to prevent under-cooling in the engine. The operating temperature in the manual is 248 degrees Fahrenheit. 

We also want to add the new radiator and get all the hoses setup for that.

### Outcomes

Here are the goals of this project:
- cool the engine to the desired temperature
- choose an appropriate thermostat
- implement the new radiator
- chose appropriate hoses and connections for the links

### Constraints

Notable constraints are as follows: 
- minimum hose distance cannot be decreased
- new radiator must be used

### Related Issues

This is under the powertrain umbrella #10 .

chadjmccolm, 2017-07-03:
The radiator was spec'd to have an inlet temperature of 180F so that's what we'll use to ensure we are within the cooling ability of our radiator within the safety factor specified in the MECH 459 report on this topic (on drive under cooling systems). 

[image]

chadjmccolm, 2017-07-03:
A thermostat was purchased from Lordco with a grooved gasket. The thermostat was a Gates 33611 and the gasket was a Gates 33708. The thermostat will be modelling approximately and heated up to measure how the length changes at and above the operating temperature chosen.

chadjmccolm, 2017-07-03:
A sketch was produced to save modelling time because of the geometry. Recommended dimensions are double underlined. Lower housing depth has yet to be decided but must be between the values specified for the fully expanded and contracted thermostat. 

[image]

chadjmccolm, 2017-07-06:
Interesting development: limitations on anti-corrosive materials was unfounded. The coolant system must be 100% water according to the rules. 

[image]

chadjmccolm, 2017-07-08:
Following today's design review these changes will be made to the model:
- move the lip to the end of where the hose attaches: increases machinability and removes redundant material
- switch to two bold design of R2 for the connector: allows easier access to attach the inlet and improves elegance of the design
- Make the ID the same for the two outlets: allows the housing to be mounted on a rod for manufacturing 
- Lock all concentric bits in the model: reduces model system use
- Match hose OD to top and bottom so the hose curve flows into the housing: aesthetic
- Make the bores tighter around the thermostat and create a straight bore on top versus the current slant: tighter is better if the machining time is the same

What we need (will be purchased tomorrow):
- G311-063 hose from Green-Line (5ft)
- 1' length of 3.5" aluminum 6061-T6 from metal supermarket
- 6x M5 x 20 flathead Torx high-tensile steel fasteners (only need 2 but it's better to keep some in stock because we will lose some eventually and they're a very generic size

isaaccormack, 2017-07-10:
R4 thermostat will include:
-thinner wall by rodney's discretion
-smaller OD of overall piece through minimizing clamping jaws size
-an elbow to hold the on thermostat piece from

smithbeuvic, 2017-07-17:
Design review with Rodney resulted in the decision to use a bolt-on insert design with O-Ring. He was reluctant to make such a large pipe thread (M20) and confirmed that straight thread and pipe tape will not seal. 

Looks like fabrication is starting tuesday morning ( July 18th ) or wednesday morning. 

Still unsure if we can trade stock material with Rodney...

smithbeuvic, 2017-07-22:
Fabrication commenced friday morning and will continue in the week of July 23rd in the Mornings. The firehydrant of aluminum stock was sucessfully traded for appropriate stock. The drawing package looks good, will update and add notes as fabrication continues.

chadjmccolm, 2017-08-02:
A sketch of the rejoiner part has been completed with some notes. @ciaraduesterwald will be modelling this with my help where necessary. 

[image]

smithbeuvic, 2017-08-02:
We would use pipe stock for this not the 1" solid stock already bought for the thermostat body.

chadjmccolm, 2017-08-02:
How would we do the ends then? Turn the rings and shrink fit them on?

smithbeuvic, 2017-08-02:
Max size of circular hole in b-b is 0.5" were not allowed to drill into or out of non flat surfaces in the shop. Will have to finish in q-hut. I might find an alternative.

smithbeuvic, 2017-08-02:
Sorry Chad. Just use NOM inner diam 0.5" and NOM outer diam 1" we can machine lip from there

chadjmccolm, 2017-08-03:
I think it looks a little funky with the 0.75 OD 0.5 ID main pipe but I don't really want to change the inner diameters where they meet because I don't want to accidentally pressurize one stream. Something we should think about though. A model is complete on GrabCAD under "2017 Formula Car\2017 Car CAD\Cooling System Rejoiner\R1 - Basic\". 

Let me know if you have any suggestions or changes and if not we can get some drawings and machine it.

chadjmccolm, 2017-08-08:
2 O-Rings were purchased because their cost was insignificant. A Green line 00093-10 was chosen to match the design spec. A M3 bottoming tap was also purchased for the insert mounting holes.

chadjmccolm, 2017-08-11:
Regarding the joiner a T junction was purchased instead of a wye. I'd have liked to use a wye but the cost benefit wasn't there. I inspected a few new KTM bikes that they all used T junctions so I'm assuming that's the industry standard. Pics of inspection in "Google Drive\Mechanical\2017-2018\Projects\Thermostat Housing\Tee Junction On New Bikes"

The water pump inlet was also a 7/8 inlet instead of a 3/4 but a 3/4 has been used in the past. It's currently impossible to get on but I'll try again with a heat gun and some lube.

chadjmccolm, 2017-09-14:
Cooling system needs to be tested with a fan on it. Testing was only done without a fan so far due to time constraints and the temperature did not stabilize. The [engine review doc](http://goo.gl/YWCBNp) can provide info on the current bench setup and the startup procedure. Please consult with me first to make sure everything checks out and you understand what's going on before starting the car.

chadjmccolm, 2017-11-17:
Testing was done with a fan and system temperature was stable.

closed 2017-11-17 by chadjmccolm