UVicFH/Home #12: 3D scanning of Engine and Exhaust

smithbeuvic, 2017-07-11:
### Context
We need an accurate model of our engine and exhaust to maximize space efficiency in our engine bay. Dr. Deschev and Michael Peirone are willing to lend us their 3D scanner
### Outcomes
Solidworks compatible 3D models.
### Constraints
 - Transport: We need to bring the scanner to Q-Hut. We do not know the size of the equipement yet. 
 - Computer hardware: Michael Peirone gave the following specifications: The requirements for the scanner are as follows: Display card：NVIDIA GTX660 or higher，Display memory：>2G，Processor：I5 or higher，Memory Storage：8G or more
### Steps to reproduce behavior
 - We must set a 3-4 day time period to work on this project as the 3D scanner can be lent for 3-4 days
 - We require someone to lend a vehicle for the task
 - We require a portable computer that meets the above requirements
 - We require members to help with 3D scanning: This may include spraypainting the components to be scanned, cleaning components, making space in the shop,

chadjmccolm, 2017-07-11:
We can use my car for whatever if we need it. I don't know of anybody with a laptop matching those specs off the top of my head but I'll start asking around. 

The engine is currently off the car in the dyno room. It would be very easy to completely detach it from it's mounts and move it wherever. Also of note is that it's light enough for two people to easily carry it around.

smithbeuvic, 2017-07-11:
Thanks! it sounds like we will set up in th eHybid shop so engine transport will be no prob

smithbeuvic, 2017-07-13:
The home made developer works well. 60-40 babypowder isopropyl alchohol

chadjmccolm, 2017-07-15:
Engine scan went well. I will be attempting to clean it up in modelling programs. Notes about the experience: 
- if the surface is too uniform it will have trouble tracking it
- graphics card demands aren't crazy; we ran on twin GTX 980s and only hit 30% use of each. 
- processor demands were very high; it ate 100% of a insane series i7 processor. 
- the program does everything in RAM and very easily took up 16GB when scanning the engine on high poly; it is recommended to have more than 16GB RAM installed or take lower-poly scans. 
- lighting made a big difference; scans done in lots of daylight worked very well but scans done in the evening were more error prone

The exhaust was not easily scannable because of the uniformity so that was not completed. Human scan trial went well and will be re-attempted at a later date with more light.

smithbeuvic, 2017-08-03:
The exhaust model seems to be accurate. It is hard to get any measurement from the real exhaust for comparison. Im getting differences < 0.125" on the dimensions I could get.

closed 2017-08-05 by smithbeuvic