UVicFH/Home #15: 3D Printer Extruder Head Cooling

smithbeuvic, 2017-08-14:
### Context
The 3D printer is nearly functioning properly. Overheating is still an issue. We must mount an additional or a larger fan to cool the printer's extruder head area.
### Outcomes
A fan and an alternate fan mounting system that better directs airflow to the extruder and work piece. 
### Constraints
Mounting system should be easy to manufacture, a 3D printed part would be good as long as temperatures are not going to melt it. 
### Related Issues
Joel from Robotics has been working on this printer. It was an abandonned and poorly engineered project from before anyones time. Joel took it from nothing to almost working perfect. 
### Steps to reproduce behavior
At your own discretion. 
 - Get data for space constraints around the extruder head and temperatures there. 
 - Review other designs and cooling solutions online
 - Decide on a design concept and a fan model, run this by Joel/Ben. 
 - CAD for manufacture, Source fan. 
 - Design review. 
 - Implement.

The cost of this project should be negligible. This project does not require project proposal or project report, however it is recommended that a short project report be kept for notes/colab.

smithbeuvic, 2017-09-09:
more issues exist with the 3D printer!
 - tension the drive belts to reduce vibration
 - replace the filiment tensioner ( gear and the preassure flange )
 - modification of the extruder head, we could follow more popular designs with the standard attachement

newmanjoel, 2018-02-06:
https://www.thingiverse.com/thing:181349
This would help solve the belt tensioning issue.

https://www.thingiverse.com/thing:2246933
would help with the cooling