UVicFH/Home #22: Reduce ETB Controller Integral Windup

chadjmccolm, 2017-09-14:
### Context

The ETB controller Arduino suffers from integral windup (or, more accurately, wind-down) when the
signal is zero. This is due to the calibration of the signal changing.

What happens is this:
• The user sets zero throttle input.
• The controller sends zero signal to the driver.
• The driver sends zero signal to the ETB.
• The ETB closes fully.
• The controller reads 4% or so from the TPS.
• The controller adds the error term to the integral (negative) which makes the integral large and
negative.
• The user changes the throttle request to 15% (for example).
• The integral term is so large and negative that it takes up to a few minutes for it to be accurate
again.

Uncaught, this _could_ cause an integer overload as the integral term becomes extremely large and
negative. Practically, the only issue is that it makes the ETB unresponsive for a short period of time.
To eliminate this error code can be added to the existing code (on Google Drive) which sets the 0 point
on the TPS based on zero initial output when the program is first run. The maximum can be calibrated
when the users inputs 100% throttle and the maximum current is given to the driver.

Other solutions may exist and are at the discretion of the designer.

Additional info can be found in the engine review doc [here](http://goo.gl/YWCBNp).

### Outcomes

The outcomes for the project are the following:

- [ ] reduce integral windup in the ETB Controller's PID scheme

### Related Issues

Related issues will be added as they are created.

tmurias, 2017-09-28:
All issues with the PID code seem to have been fixed by limiting the speed of the main loop. This prevents the dt terms from becoming too small, which was making the integrals and derivatives inaccurate.

closed 2017-09-28 by tmurias

chadjmccolm, 2017-09-28:
This was done by adding a 10ms delay during the Arduino's loop.