UVicFH/Home #26: Wheel Covers

smithbeuvic, 2017-10-15:
### Context

Our tire covers need to be remade or altered to fit appropriately. These preserve the tires and look good. 

### Outcomes

 - Alterations to the current wheel covers such that they completely cover the wheels and can be put on/removed with ease.
or
 - New wheel covers made from an aesthetically pleasing but durable material such that they completely cover the wheels and can be put on/removed with ease.

### Constraints

 - Budget is $50
 - Car must be able to roll around with them on
 - Must look Awesome

### Steps to reproduce behavior

 - Measure the dimensions required for the tire sleeve
 - Decide on a good securing/tightening mechanism
 - Decide what material will be used
 - Decide how these will be manufactured
 - Buy everything and make everything

chadjmccolm, 2017-10-16:
Here's my take on it. Unsure of material. I'm against using a pull cord for tightening because I don't want it to get in the way but don't like the current attachment with velcro because I don't want the tires sitting on a hump for extended periods of time in transport etc. so I suggest a elastic on the inside only. I think filling the outside edge also gives us a good place to put some sort of eye candy. 

[image]

chadjmccolm, 2017-10-16:
Just realized that the inner face needs to change to something similar to the body in shape so that it can slip over the tire. How far the elastic we use stretches will determine how wide it is.

smithbeuvic, 2017-10-20:
[image]
 Like this one piece design but with cinching cord instead of elastic

chadjmccolm, 2017-10-20:
Yeah. Either with elastic or cord to tie up the back. Otherwise if we want to show off how cool our wheels are we can do a similar pattern to the one we have now but I'd rather the part along the tire be sewn than velcrod. Then we can do a set of grommets along each side or wearing but I think it'll be easier to build and look cooler with grommets and tied up with Paracord or something

chadjmccolm, 2017-10-20:
I actually really like the idea of having it open faced front and back like it is now and tying with coloured cord. We could do a blue white scheme between the cord and the fabric

smithbeuvic, 2017-10-20:
fully agree, I like the wheel showing.