UVicFH/Home #28: Noisy TPS signal with Fuel Pump Running

chadjmccolm, 2017-11-02:
### Context

When the fuel pump turns on the TPS signal get's noisy and the range changes. This makes controlling the TPS nearly impossible. A voltage drop is observed from 12 to 10V at the Microsquirt.

### Outcomes

Clean Up Signal to TPS and let TPS controller work effectively

### Constraints

We would rather not replace the fuel pump but with that wine we may have to. 

### Steps to reproduce behavior

Plug in the Arduino controlling the TPS and open Arduino software on the computer to monitor the serial port. Serial plotter is helpful. 

Power the ECU etc. off the main switch and observe the ETB's function. Now power on the fuel pump and observe the signal. 

### Suggestions

As discussed in the meeting there are a number of reasons why this could be happening. Off the top of my head: 
- the draw may be loading the power supply such that the actual voltage is fluctuating and causing the sensor values to fluctuate
- there may be induced EMF due to the electric motor in the fuel pump

If it was the first one, the problem would be fixed by running the fuel pump off a different 12V source. Not it may draw up to 6A continuous. 

If it was the second one, you could fix it with shielding the wire and grounding the shield OR you could change how much voltage is induced by altering the path the wires needs to take from the sensor to the Arduino.

chadjmccolm, 2017-11-09:
This is likely due to RFI travelling through ground and through the air. The solution would be to measure the interference produced for dominant frequency using the scope on the fuel pump and fit a capacitor on the way to ground to block those frequencies. All signal wires should also be shielded to protect from extra interference.

UVicFormulaHybrid, 2017-11-14:
Is it possible to test the TPS signal noise with the fuel pump grounded to the building opposed to the engine trolley

chadjmccolm, 2017-11-14:
The pump was tested with a scope. 

> "Hey I went in and checked it out. Your suspicion is right. There’s about a 3-6 V swing when the pump is on. The frequency was hard to measure but it looks like around 500-1000kHz" - Robert Lee

This can be dampened by using a DC filter which are available on Digikey. For now we will run the fuel pump off a separate power supply which is a more elegant solution than having 12V run to the power supply and ground run to the building because although the building should be grounded the ground voltage may be different than the power supply ground.

chadjmccolm, 2017-11-24:
Because we are using a DC switching power supply this interference is likely being caused by the switching and the internal capacitance not being high enough. Oliver ran into this same problem. The recommended solution is to run the bench off the batteries because batteries have a high internal capacitance. Or use a large capacitor on the output of the power supply.

chadjmccolm, 2018-01-15:
Adding battery capacitance solved this issue.

closed 2018-01-15 by chadjmccolm