UVicFH/Home #32: Telemetry

chadjmccolm, 2017-11-16:
### Context

We want to send the car's data to a server and display it on a webpage to reduce the amount of information clogging up the driver's display and allow for easier data logging. 

We have previously purchased parts for this:
https://www.ubnt.com/airmax/rocketm/
https://www.ubnt.com/airmax/nanostationm/
https://www.ubnt.com/airmax/airmax-omni-antenna/

The data will be sent via the MQTT protocol to a server created by @BrendonEarl. 

The server is powered by NodeJS. The data will likely be sent by an Arduino running an MQTT client like [this one](https://pubsubclient.knolleary.net/).

The flow will be as shown here: 
[image]

### Outcomes

- [ ] Create a link between the car and the server
- [ ] Setup a MQTT server
- [ ] Configure the Arduino to read CAN and send via MQTT over Ethernet
- [ ] Create a web page to view the data recieved

### Constraints

We should use the technology we've already purchased.

MrMikeGrant, 2017-11-16:
With regards to testing the telemetry system, I have been making progress with a HIL simulation. Essentially the MABX will be wired up to a dSPACE simulation platform and receive signals in real time that it would normally see on the car. If we wire up the telemetry system as shown above, we could test and validate its implementation.

chadjmccolm, 2017-11-17:
Arduino in the above schematic will be done instead with Raspberry Pi which will also power the dashboard. What is displayed on the dash is up for debate but I think what we really need is:
- rpm
- speed
- area for warning lights (overtemp, etc.)

Everything will be sent to the MQTT server using a python library and the dash will also be powered the same Python program. 

This also opens up the possibility of sending voice data back and forth for the driver to communicate with the pit. This year, we likely won't have time to do this but it's a good future plan.

chadjmccolm, 2017-11-19:
Here is my idea for the dash. Working on building the GUI currently. I've decided to use Python because it has great libraries for MQTT and CAN communication. 

[image]

So far the pre-reqs for getting this running on the RPi are:
- python 2.7
- pyside

The program will run in fullscreen by default and the Pi should be configured to run this program on boot. 

pyside can be installed with pip with the commend `pip install pyside`

chadjmccolm, 2017-11-19:
I wrote a summary of the telemetry system for the Whiton Fund which is available [here](https://goo.gl/iYLLQ8)

Good resource for future reference.

chadjmccolm, 2017-12-01:
A fuel and charge level indicator will be placed on the UI on either side of the current information or maybe below it. Haven't decided yet. The libraries will be switched to PyQT and python 3 to correspond with the CAN library I want to use, python-can. The QML file controlling the GUI will likely remain the same, the only changes will be to the python application itself.

BrendonEarl, 2017-12-03:
@chadjmccolm with regards to the display: I'm not sure if you were previously aware of this, but part of the motivation for the telemetry system was so that we could simply the dash and no longer use the display as it was hard to see on bright days. While it does look super sexy, it wasn't super functional when racing. We thought a bright LED array would be more noticeable as a tach.
Grayson previously had a design, which should be on grabCAD or the drive somewhere, for a new dashboard. Not that you guys should definitely drop the screen, but it may be work exploring

tmurias, 2017-12-04:
@BrendonEarl The dash design Grayson and I worked on last year is in the Electrical/2016-2017/Dash folder in the drive. I was going to change some parts of it this season but put it on the back burner since I'm spending all of my time on the AMS, and since Chad wants to combine the dash with telemetry I figured we'd just abandon it. It's still there (and pretty much complete) if we change our minds, but it might be nice having the dash and telemetry combined. As long as the new LCD will be bright enough and we only display the really important stuff, it should be fine.

chadjmccolm, 2017-12-04:
The plan is to ditch the dash next year and improve it this year. I'm almost done recreating the UI using the PyQT library. Just gotta add fuel and battery level and then I'm Gucci. I've also switched all the graphics to SVGs and did all the positioning relational so as long as the aspect ratio is somewhere near 16:9 it should look great regardless of screen size and pixel density. I'm also probably going to make a shroud for it of sorts to block out light. 

Design note: I've changed the tach from a linear with torque curve to a radial because demoing the operation before showed me that our gear ratio is too close for the lowered resolution of a linear tach and the larger sweep of the radial tach is the best idea.

chadjmccolm, 2017-12-06:
Version 2.1 of Teledash complete. Preview available here: https://drive.google.com/file/d/1Qh-QPWryCmi-VmfAidZ1QDS9iwBabsGg/view?usp=sharing 

Fuel and battery has been added with a backdrop so you can see where 100% after they've started going down on suggestion of @smithbeuvic

chadjmccolm, 2017-12-08:
Mosquitto is going to be our MQTT broker. If it runs on a windows system it needs to be installed with some dependencies. When installing them, INSTALL OPENSSL < V1.1 OTHERWISE IT WON'T WORK. Learned that the hard way.

chadjmccolm, 2018-01-12:
Installation instructions for dash code on windows system: 
- Get the latest version of [Python3](https://www.python.org/downloads/) (currently 3.6.4)
    - Make sure to select "Add Python to Path" option in installer
- Verify `python` and `pip` commands are available from the command line
- Install pyqt5 from wheels with `pip install PyQt5` from the command line
- Install paho-mqtt with `pip install paho-mqtt` from the command line

Now the code can be run. 
(this will likely be modified soon)

Clebeuf, 2018-01-14:
@chadjmccolm: Reminder about the list of data points for the telemetry dashboard.  A general list would be great for now, but eventually it would be great to sit down and create a prioritized list of data needed for each use case for the telemetry dashboard.

Once we have an idea of the data (and it's) general structure we can get started on the front end of the telemetry dashboard. Then hopefully when the car is ready to transmit it, we can hook it up pretty easily with a little tweaking.

chadjmccolm, 2018-01-15:
A whole lot of dependencies downloaded later, the program is now working on the Raspberry Pi 2 running Ubuntu Mate just the same as it works on Windows. I'm going to remove a bunch of unneccesary programs from the OS and then create an img for reference.

chadjmccolm, 2018-01-15:
Here's how to set up mosquitto on windows so that we can run code that simulates signals coming in from the Pi. 
1. Download [mosquitto](https://mosquitto.org/download/).
2. Install Mosquitto. Don't worry about the dependencies right now. Installation may appear to not complete or fail. 
3. Download and install [Win32 OpenSSL v1.0.2n Light](http://slproweb.com/products/Win32OpenSSL.html). Install the dlls to the OpenSSL/bin folder. We need to grab them later. 
4. Download [Pthreadvc2.dll](http://www.sourceware.org/pub/pthreads-win32/dll-latest/dll/x86/).
5. Copy libeay32.dll ssleay32.dll from the OpenSSL32(/bin) folder and pthreadVC2.dll into the mosquitto installation directory. 
6. If you don't already have Visual C++ Redistributable Packages for Visual Studio 2013 x86 you will need to download and install that as well. 
7. Reinstall mosquitto so that it can use those DLLs to make itself a service. 

At this point, I would go into services (run services.msc) and change the start type of the mosquitto service to manual so that you can start and stop it as required.

chadjmccolm, 2018-01-17:
@Clebeuf here is the data that I can supply the telemetry dashboard as far as I know. Not sure what we're getting from the accumulator yet because we don't have one yet. 

https://docs.google.com/document/d/1b_Rnik_16HKZxufBX0N3VIh2z4FQfbZCLo3LWnhIgOw/edit?usp=sharing

Stored in our Drive in Drive\Software\2017-2018\Signal Hierarchy.gdoc