UVicFH/Home #36: Documenting Motor Controller

chadjmccolm, 2018-01-16:
### Context

We don't have a solid documentation for how the motor controller works with the rest of the car and I would like to know: 
- what inputs it needs
- what current those inputs will have
- what outputs it gives via CAN or other signals

Our controller is a Kelly Controls KDH14601E. All information can be found online as it is still an active product. 

### Outcomes

The following are desired outcomes of this investigation: 
- pinout of the controller as it applies to our car
- wiring diagram in Altium
- data on CAN signals added to main spreadsheet

### Related Issues

This will come together with #17 Wiring Schematic Cleanup to increase the validity and accuracy of the wiring harness so that we have 0 issues down the road.

kurtiswoods, 2018-01-17:
[KellyKDHEUserManual.pdf](https://github.com/UVicFH/Home/files/1638061/KellyKDHEUserManual.pdf)

Pages 8,9,10 are the pin diagrams explaining the output, input, signals, and currents for the J1 and J2 pins

kurtiswoods, 2018-01-17:
[image]
[image]
[image]

kurtiswoods, 2018-01-17:
And this might already be known but I’ll throw it in anyways  for the RS232 port — “Configurable and programmable with a host computer though RS232 or USB. Provide free GUI which can run on Windows XP/2000, Windows 7 and Vista(recommend using Kelly Standard USB To RS232 Converter).“

UVicFormulaHybrid, 2018-02-07:
I have unearthed this email confirming that our controller is rated for 500VDC: https://docs.google.com/document/d/1H4KAorWFNJrISrr2Zw_n9ERj4ZBQ4eDVmgzVubcDKrc/edit?usp=sharing

chadjmccolm, 2018-02-23:
Motor controller must be wired into the car as follows for forward only motion:
J2: Pins 1 and 2 for power and ground must be given 12V. Pin 5 must be hooked to a 0-5V analog signal for throttle. Pin 6 must be hooked to a 0-5V analog signal for brakes. Pin 9 is used to activate the brake input. 
J1: 10 and 11 must be hooked to CAN high and low.

This has been tested and function confirmed.

closed 2018-02-23 by chadjmccolm